pyroesp/PlayStation-1-Reset-Mod #4: Guide to know where to solder each point

pyroesp, 2019-10-16:
At the moment, the only guide there is is in the pictures folder and is for a PU-18 motherboard.  

Either make a guide for :  
- Where to solder on all different PlayStation 1 motherboards.  
- How to find the location to solder the mod onto.  

**1st option:**  
- Clear location with pictures of all different boards  
- Need outside help or donor boards to get this  

**2nd option:**  
- Very basic multimeter skill necessary  
- No need for outside help or donor boards  
- Prone to user error

pyroesp, 2019-10-16:
From the 2nd option and outside help, option 1 could be made...

pyroesp, 2019-10-16:
For option 1, pictures and annotation of solder points are needed for following motherboards:  
- PU-7  
- PU-8  
- PU-18  
- PU-16  
- PU-20  
- PU-22  
- PU-23 ?

Damn-Deal-Done, 2019-11-25:
Hello.

I have made guides for all these now (PU-8/PU-18/PU-20/PU-22). Easy to follow high resolution images. I worked most of today taking apart my PlayStations, taking high resolution photos, and probing with my multimeter. I just want to neaten up the images and I can share them with you. Probably tomorrow.

Is your board layout finalised now?

Are you able to release a Hex with all the fuses pre-programmed for programmers like the TL866?

Would it be possible for you to send me a board for testing?

Much appreciated.

Damn-Deal-Done, 2019-11-26:
Hi.

I have uploaded them to my Github. https://github.com/Damn-Deal-Done/playstation-in-game-reset-solder-points

They have my branding. Let me know if you want some without.

Not tested the guides, but all the points have continuity to the correct controller pins.

I can get a few made and test all these installs, but only once your board is finalised.

Cheers

pyroesp, 2019-11-28:
Those look really good! Awesome, thanks!
I'll make a copy off the images to put in my repository, so I have them too.
I'll have to add Credits to my readme :D 
Will do this later today.

Thanks again.

Damn-Deal-Done, 2019-11-28:
No problem. I can do PU-23 later today. I don't have PU-7 or PU-16 but will do them if they cross my path for an install.

Once the chips arrive in a week or two, I will test and work out the best placement for the modboard and update the images.

Got my PICkit4 coming today.

pyroesp, 2019-11-28:
I have a PICKit3 but it shouldn't be too different than a PICKit4 so if you need any help with it, you can PM me on twitter over the weekend or in the evening in the week and I'll help you get setup.

Damn-Deal-Done, 2019-11-28:
Cheers. I think I managed.

Installed MPLAB X IPE and everything was straight forward.

Programmed the chip in an adapter plugged into a breadboard. Powered chip with the PICkit4. It was detected, successfully programmed and verified. :)

[image]

hurda, 2020-04-24:
I made a guide for PU-23 using a service-manual, but the mainboard-pics I used are from Wikipedia.
https://commons.wikimedia.org/wiki/File:Sony_Playstation_1_SCPH-9002_motherboard_top.jpg
https://commons.wikimedia.org/wiki/File:Sony_Playstation_1_SCPH-9002_motherboard_bottom.jpg
I don't know how to properly follow the requirements of the CC-license there.

pyroesp, 2020-04-24:
Hey thanks for that, no worries.
I know this mod works on all PS1 but I don't know if it's really useful to have it on a PS1 that doesn't have a parallel port / PSIO device.

If you want to post the pictures, add a link to CC BY-SA 4.0 license, say it's from wikipedia and add the link to the wiki images. I think you also want to say you modified the image. 
That's pretty much it I think.

Damn-Deal-Done, 2020-04-24:
I made a PU-23 a few months ago. Just uploaded it. By all means make one if you still want. But mine is here and it matches the others. https://github.com/Damn-Deal-Done/playstation-in-game-reset-solder-points. No license needed.

Handy for rage quitters. :)

Cheers

hurda, 2020-04-25:
> I made a PU-23 guide a few months ago. Just uploaded it. By all means make one if you still want. But mine is here and it matches the others. https://github.com/Damn-Deal-Done/playstation-in-game-reset-solder-points. No license needed.

Ok, yours is better. :)
Except for one thing: There's a nice soldering-testpoint for DAT just below the controller-connector left of the "CN102"-marking. The rightmost one in the group of three.
Or at least that one should be for DAT according to the service-manual.

Damn-Deal-Done, 2020-04-25:
Cheers. Will have a look. I checked them all for continuity, but I have the most garbage multimeter imaginable.

Damn-Deal-Done, 2020-04-27:
Cheers, that was correct. Image is now updated using the test point for DAT. I am officially blaming my tools.

hurda, 2020-04-28:
You uploaded the old image again. ;)

Damn-Deal-Done, 2020-04-28:
That is correct. I need to start getting more sleep. Reuploaded.

pyroesp, 2020-12-21:
PU-23 has been added, finally.

closed 2020-12-29 by pyroesp