pyroesp/PlayStation-1-Reset-Mod #6: Printing PCB

ijacquez, 2019-10-18:
Would these be the right options to print a few PCBs?

[image]

pyroesp, 2019-10-19:
I don't have all those options when I go for the PCB prototype quote.

This is what I'm going for : 
[image]

pyroesp, 2019-10-19:
Make sure you get the latest version which will hide the JLCPCB order number.
Also, the resistor is gone as it's not needed due to internal pull-up, see #5 .

ijacquez, 2019-10-20:
Thanks, I'll update.

The 3.5V has changed as well. Will a part list be given as well?

pyroesp, 2019-10-20:
I'll either add the part list to the readme or add it to a wiki.

pyroesp, 2019-10-20:
I've added a part list to the readme after the schematic and pcb images.

You need anything else ?

ijacquez, 2019-10-20:
Thank you. I'll order a batch and make a few to test them out.

I guess for future revisions, it would be nice to have a connector of sorts for the PICkit3

I'll have to see what motherboard I have.

I have a PS3 to PSX adapter, so I'll see if it works.

Thanks.

pyroesp, 2019-10-20:
@ijacquez : A connector would add cost for a one time thing.
Best way I see how to do it is make your own cable with pogo pins, that's how I'm going to make it.

If the PS3 controller works on the PS1 with that adapter, then that adapter must be "translating" the communication to the PS1 protocol, which means that it should work if the device ID is one of the three programmed in the code.

ijacquez, 2019-11-10:
Got the PCB, soldered the chip, the cap.

Programming required that I power it through USB (v2.3). That threw me for a loop. I can only at this point erase, program, check blank, and verify. All passed.

What is a good way to test this without directly connecting to the PlayStation?

[image]

pyroesp, 2019-11-12:
Yes for programming, you need to supply the board with voltage.
You'd do that from the pickit 3 programming menu. Probably something I need to add to the readme!

If you want to test the board I would suggest making a test program for it. 
There is no way to test the board out of the PlayStation unfortunately.

ijacquez, 2019-11-12:
I have made a fairly big mistake with programming the chip.

So when the board needs to be supplied with voltage, does that mean I simply connect the PICkit3's VPP to VDD, then within the programming menu, have the PICkit3 supply the voltage, and set the voltage to ~2.5V?

pyroesp, 2019-11-14:
No, the VPP of the PICKit3 goes to the VPP/MCLR pin on the back. VPP outputs a higher voltage than VDD when programming, which potentially could damage the PIC microcontroller!

The silkscreen on the back should mention this :
[image]

In the advanced mode of MPLAB IPE go to the Power tab. These are my parameters for the 16F18325:
[image]
These parameters should have a default value for the 16F18325, but the thing you might have to check is the usage of high voltage programming and also that checkbox for powering the target with the PICKit3.
The VDD voltage can be either 5V or 3.5V, it doesn't really matter as long as you're programming the PIC when it's not connected to the PlayStation.

Once it is programmed and verified by MPLAB IPE, then you only need to install it in the PlayStation.

pyroesp, 2019-12-14:
@ijacquez hello, just letting you know that there's a new firmware version available.
https://github.com/pyroesp/PlayStation-1-Reset-Mod/releases/tag/v1.1

There was an error while making the PCB where data and cmd were switched. The new firmware fixes this.
Gerber files are still the same as in v1.0

ijacquez, 2019-12-15:
Thanks for the heads up. I'll give the project another whirl.

pyroesp, 2020-02-14:
Since there's nothing new here I'll close this issue.

closed 2020-02-14 by pyroesp