pyroesp/PlayStation-1-Reset-Mod #13: Alternative button-combo for longer reset

ArcadeTV, 2020-11-24:
Thank you very much for this great project, great documentation and dedication to this!

I would like to suggest adding a second button-combo to the code, which when triggered holds the reset for appx. 2 seconds. 
The reason for this is the release of the X-Station optical drive emulator, which is responding to the longer reset and returning to the menu instead of resetting the game. There's a IGR for the snes which works as described when using an sd2snes/fxPAKpro.

pyroesp, 2020-11-24:
Looking at the xStation doc it looks like it needs "about 3 seconds" to go back to the menu.
I could definitely add something like that.

ArcadeTV, 2020-11-25:
On the SNES the IGR uses a similar combo:
(https://github.com/borti4938/SNES_MultiRegion_with_DeJitter_QID)

simple reset: L + R + Select + Start
double reset (back to main menu of sd2snes): L + R + Select + X

Since the layout of the PS1 controller is similar to the snes, I'd suggest this combo for the change.

Thank you very much for even considering adding this, it's very much appreciated!

pyroesp, 2020-11-25:
> ...
> double reset (back to main menu of sd2snes): L + R + Select + X
> 
> Since the layout of the PS1 controller is similar to the snes, I'd suggest this combo for the change.
Yeah, I was thinking about using the x button too, instead of START, for the longer reset. 
Press x for xStation reset. It's the perfect button for it.

> Thank you very much for even considering adding this, it's very much appreciated!
No worries! When I talked about the xStation online and compatibility with my mod, people told me that they'd most likely also buy the PS1Digital so mine wouldn't be needed.  
From then on I never really paid much more attention to this, but I'm not against adding features like this to the mod.  

It only takes a minute to change the code, but I'll have to test this to see if it works as intended.
Hopefully I can spare some time to do this this weekend.

If not then you'll see it when I close this issue and/or post about it online.

ArcadeTV, 2020-11-25:
No stress, take your time. I can test too once you pushed the changes. I might have a PIC16F18325 laying around.
Thank you very much again for this!

pyroesp, 2020-11-30:
After a bit of testing I've commited the new code for the PIC microcontroller.
The xStation should be supported now.

492eb7d

closed 2020-11-30 by pyroesp

Damn-Deal-Done, 2020-11-30:
ArcadeTV if you want to test a board with the new firmware just let me know and I can send you one for free. It's not a problem.

ArcadeTV, 2020-11-30:
> ArcadeTV if you want to test a board with the new firmware just let me know and I can send you one for free. It's not a problem.

That's so nice, thank you! I have a breakout board for testing, also ordered a pcb with my latest jlcpcb order ;-)