pyroesp/PlayStation-1-Reset-Mod #15: Programing the PIC16F18325

mroy1300, 2021-11-20:
Hello, 

I'm trying to program the PIC16F18325 with the supplied hex file but when I read the device memory the hex that gets output is different from the original. Am I doing something wrong? How are you supposed to verify that the chip was successfully programed?

My apologies if this is not the best place to post this but I didn't know where else to do it.

Let me know if you need any additional info, I didn't know what info I should supply.

pyroesp, 2021-11-21:
It should be identical to the hex. I'm not doing anything fancy to program it.
You can check the "how to" in the wiki.

Also try another 16F18325 just in case that one is doing weird things.

mroy1300, 2021-11-21:
I have been doing more testing and I found that a fresh non-programed chip output the same hex file as a programed chip. I'm wondering if the programmer is either not programing the chip or if I'm reading the hex file wrong.

I have attached pictures of the chip, board, and programer. The VDD, VPP, CLK, VSS, and DAT pads are out of order on the board so I made a jumper to correct that.

I also took a video of the process. It looks like the hex file is not being written to the chip. I'm pretty good with computers but this is my first attempt at programing microchips. Hopefully I'm just doing something wrong.

[image]
[image]
[image]

https://user-images.githubusercontent.com/94731368/142781816-cbd6bbb6-b071-4ef7-b262-45f8858fef55.mp4

pyroesp, 2021-11-22:
When programming through IPE, you can see the output saying "Program/Verify completed". Which means it's programmed and verified the contents of the chip.
If you want to read the data back, you can just click on the read button instead of going to the IDE.

Also, you are selecting EE/Flash data instead of device memory in the IDE.

mroy1300, 2021-11-22:
I see the log saying it's verified but the hex doesn't change. I have read Device memory through IDE also but I get a huge hex file with 1061 lines so I figured that was reading the memory from the programmer. 

When I read through the IPE I don't get any output besides the IPE saying it was read. DO I need to turn on logging? 
[image]

My apologies, I assume I'm doing something wrong. I just can't figure it out.

mroy1300, 2021-11-23:
OK, I have it so that I can read and export the hex file in IPE. When I program the chip everything looks right and I even ran a blank check on the chip. The program confirmed that the chip is not blank but when I read the hex file it has not changed. I also confirmed that the board does not work when installed into the PS1. I'm at a loss at this point. Everything seems like it should be working.

pyroesp, 2021-11-23:
I suggest you try programming a blinky LED to another microcontroller to see if your setup works.

closed 2021-12-04 by pyroesp