pyroesp/PlayStation-1-Reset-Mod #18: IGR for both Controller Ports

GeAd23, 2024-02-15:
Is it possible to use two IGR boards for Controller 1 and 2. How can I do this ? (Modchip is installed too)

I would use one igr from (flat) unleashedmods and a normal igr board. 

Yours Sincerely

pyroesp, 2024-02-19:
Hi sorry for the late reply.
Each port has it's own ATT (or SS) signal as seen here on port 1:
https://raw.githubusercontent.com/pyroesp/PlayStation-1-Reset-Mod/master/pictures/guide/playstation-controller-pin-map.jpg

IGR1 should connect to ATT1 and IGR2 should connect to ATT2 if I'm not mistaken.

GeAd23, 2024-02-19:
Okay. Thank you. The other ports are connected together Else SS ? Like SPI work. 
I would test it out. 

Yours Sincerely

closed 2024-02-21 by pyroesp